nordicsemi/Android-nRF-Mesh-Library #537: Problem with IV index udpate

Fabbros, 2022-10-10:
Hi, 
I did some test with IV index using nRF Mesh, here what I did:
1. provisioned 2 devices in a new mesh (device that have nordic SDK, IV test flag enabled)
2. from one devices, I triggered the IV index update from 0 to 1
3. after the finish of the IV procedure, I tried to interact with devices using nRF Mesh -> OK (get/set status, bind app key etc)
4. from the same device of point 2, I triggered the IV index update again, from 1 to 2
5. after the finish of the IV procedure, I tried to interact with devices using nRF Mesh -> FAIL (can't get/set status, can't bind app key etc)
6. I also tried to add a third device into the same mesh -> provisioning ok but every next operation fails (can't bind app key, get/set status is not possibile due to missing app key binding)

It seems that when mesh IV is over 1 nRF mesh can't interact with it because app still use IV 0 while the devices use IV 2.

I tried the steps above using:
- Android nRF Mesh v3.2.4
- iOS nRF Mesh v3.2.0

philips77, 2022-10-11:
Have you toggled the "IV Test Mode" switch in nRF Mesh in Settings?

Fabbros, 2022-10-19:
Is there anyone who can help me?

philips77, 2022-10-19:
Hi, sorry for late reply.
Do you have that issue on both Android and iOS?
Our apps are getting the IV index from Secure Network Beacons. Try disconnecting the phone after each IV update, do that the phone could get the current version. It should be sent over GATT, but perhaps it's not. But for sure it will be when you reconnect to a proxy node.

philips77, 2022-10-20:
Let me be more clear here.

During an IV Update procedure, the Proxy Node should rely Secure Network Beacons with updated IV state to the phone. However, seems like that is not the case. As SNBs are sent over ADV bearer regularly, all the other nodes get the current IV Index state almost immediately, but the Proxy Node filters them out and only sends them on new connection and when IV changes. Could you check if you get it retransmitted from the Proxy over GATT bearer?
If not, after each phase transition in IV Index update disconnect and reconnect the phone so it gets the SNB on reconnection.

Fabbros, 2022-10-20:
Hi Philips,
I have the issue on both, Android and iOS.

Can you please explain how to check if the SNB is retransmitted?
BTW, after a reconnect, app still can't communicate with provisioned devices.

philips77, 2022-10-20:
Steps to test:
1. On a new brown provision a node.
2. Enable IV Index Test Mode in both the phone and the node.
3. Make sure the phone is connected to the node.
4. Trigger IV Index update from the node to IV=0, mode=active.
5. A SNB should be received in the phone:
[image]
6. You should be about to send messages.
7. Trigger IV Index update again. This time it will be IV=1, mode=normal.
8. Verify that the SNB was received and handled, repeat sending and receiving messages to test.
9. Trigger update again, to IV=1, mode=active.
10. Repeat testing.
11. ...

Make sure the SNB was received in each step. If not, the following will be rejected it required time hasn't passed:
[image]

Fabbros, 2022-10-24:
Hi Philips, 
I ran your steps without being able to get info from the devices in the end.
IV index is 2 and in normal operation.
[image]

I tried a reconnect:
[image]

Fabbros, 2022-10-24:
Can you tell me how to better filter log obtained with iOS Console?

On Android, from terminal, I used the following:
adb logcat --pid=$(adb shell pidof -s no.nordicsemi.android.nrfmeshprovisioner)

philips77, 2022-10-24:
> I ran your steps without being able to get info from the devices in the end (Android).
IV index is 2 and in normal operation.

Again, so is it working, or not? What would you expect instead?

Looks like in both listings you're getting a valid SMB with IV = 2 (normal operation).

philips77, 2022-10-24:
In the Console app you may filter by SUBSYSTEM: com.nordicsemi.nRFProvisioner

[image]

philips77, 2022-10-24:
Also, make sure you enable Info and Debug messages:
[image]

Fabbros, 2022-12-12:
Hello again

I ran a lot of tests in the last few days following your indications.

### iOS app
iOS nRF Mesh seems to work almost fine, I was able to reach IV 4 and still was able to read/set a GenericOnOff state from the device. However, I think there is an issue: when stepping to the next IV the app gets the SNB indicating each step (first the SNB indicating IV n+1 and active state and later the SNB indicating IV n+1 with normal operation) but can't send messages until a disconnect+reconnect to the proxy GATT. I get the same issue when provisioning a new node: the new node gets provisioned with the correct IV (for example IV 4) but then the app can't get the composition data until a disconnect+reconnect.
Step to reproduce are:
1. provision a few nodes (with the IV test flag enable of course)
2. step the IV index (until 4 for example) from the nodes
3. power off the nodes already provisioned
4. provision a new node -> the app will correctly provision with IV 4 but then can't get the composition data until a disconnect+reconnect to the proxy GATT

Here I gathered a bunch of logs retrivied while testing with IV updates with iOS

**GenericOnOffServer get of a node -> OK**
com.nordicsemi.nRFProvisioner	11:32:44.327605+0100	nRFMeshProvision_Example	Sending GenericOnOffGet() from: Primary Element (0x0001), to: 0006
com.nordicsemi.nRFProvisioner	11:32:44.327764+0100	nRFMeshProvision_Example	Sending Access PDU (opcode: 0x8201, parameters: 0x)
com.nordicsemi.nRFProvisioner	11:32:44.332606+0100	nRFMeshProvision_Example	Sending Upper Transport PDU (encrypted data: 0x5CB5, transMic: 0xCF676216) encrypted using key: App Key 1 (index: 0)
com.nordicsemi.nRFProvisioner	11:32:44.332969+0100	nRFMeshProvision_Example	Sending Access Message (akf: 1, aid: 0x2B, szmic: 0, data: 0x5CB5CF676216)
com.nordicsemi.nRFProvisioner	11:32:44.334290+0100	nRFMeshProvision_Example	Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 5, seq: 36, src: 0001, dst: 0006, transportPdu: 0x4DA00D95D8BFB6, netMic: 0x1FAEEC0D) encrypted using Primary Network Key (index: 0)
com.nordicsemi.nRFProvisioner	11:32:44.334923+0100	nRFMeshProvision_Example	-> 0x0062A0A07AF4A489DC534DA00D95D8BFB61FAEEC0D
com.nordicsemi.nRFProvisioner	11:32:44.689084+0100	nRFMeshProvision_Example	<- 0x00628D419C72D8CDDD6F65F05DBAB5954A053CFB844BBBA77D
com.nordicsemi.nRFProvisioner	11:32:44.690429+0100	nRFMeshProvision_Example	Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 3, seq: 17, src: 0006, dst: 0001, transportPdu: 0x65F05DBAB5954A053CFB84, netMic: 0x4BBBA77D) received
com.nordicsemi.nRFProvisioner	11:32:44.694252+0100	nRFMeshProvision_Example	Segmented Access Message (akf: 1, aid: 0x2B, szmic: 0, seqZero: 17, segO: 0, segN: 0, data: 0x71BE6EC73E1B37) received (decrypted using key: Primary Network Key (index: 0))
com.nordicsemi.nRFProvisioner	11:32:44.694333+0100	nRFMeshProvision_Example	Access Message (akf: 1, aid: 0x2B, szmic: 0, data: 0x71BE6EC73E1B37) received
com.nordicsemi.nRFProvisioner	11:32:44.694643+0100	nRFMeshProvision_Example	Upper Transport PDU (encrypted data: 0x71BE6E, transMic: 0xC73E1B37) received
com.nordicsemi.nRFProvisioner	11:32:44.694710+0100	nRFMeshProvision_Example	Sending ACK (seqZero: 17, blockAck: 0x00000001)
com.nordicsemi.nRFProvisioner	11:32:44.694775+0100	nRFMeshProvision_Example	Response Access PDU (opcode: 0x8204, parameters: 0x00) received (decrypted using key: App Key 1 (index: 0))
com.nordicsemi.nRFProvisioner	11:32:44.695172+0100	nRFMeshProvision_Example	GenericOnOffStatus(isOn: false, targetState: nil, remainingTime: nil) received from: 0006, to: 0001
com.nordicsemi.nRFProvisioner	11:32:44.695259+0100	nRFMeshProvision_Example	Local Generic OnOff Client model on Primary Element (0x0001) not bound to key: App Key 1 (index: 0)
com.nordicsemi.nRFProvisioner	11:32:44.695393+0100	nRFMeshProvision_Example	GenericOnOffStatus(isOn: false, targetState: nil, remainingTime: nil) already decoded as GenericOnOffStatus(isOn: false, targetState: nil, remainingTime: nil)
com.nordicsemi.nRFProvisioner	11:32:44.696822+0100	nRFMeshProvision_Example	Sending Network PDU (ivi: 0, nid: 0x62, ctl: 1, ttl: 5, seq: 37, src: 0001, dst: 0006, transportPdu: 0xEDDAF1977C1583, netMic: 0x2DF580A907DB5C19) encrypted using Primary Network Key (index: 0)
com.nordicsemi.nRFProvisioner	11:32:44.696923+0100	nRFMeshProvision_Example	-> 0x00628B9DB0CF88A54034EDDAF1977C15832DF580A907DB5C19

**GenericOnOffServer set of a node -> OK**
com.nordicsemi.nRFProvisioner	11:33:18.944694+0100	nRFMeshProvision_Example	Sending GenericOnOffSetUnacknowledged(tid: Optional(147), isOn: true, transitionTime: nil, delay: nil) from: Primary Element (0x0001), to: 0006
com.nordicsemi.nRFProvisioner	11:33:18.944761+0100	nRFMeshProvision_Example	Sending Access PDU (opcode: 0x8203, parameters: 0x0193)
com.nordicsemi.nRFProvisioner	11:33:18.946157+0100	nRFMeshProvision_Example	Sending Upper Transport PDU (encrypted data: 0xCFFC4DAE, transMic: 0x84139A9B) encrypted using key: App Key 1 (index: 0)
com.nordicsemi.nRFProvisioner	11:33:18.946350+0100	nRFMeshProvision_Example	Sending Access Message (akf: 1, aid: 0x2B, szmic: 0, data: 0xCFFC4DAE84139A9B)
com.nordicsemi.nRFProvisioner	11:33:18.954992+0100	nRFMeshProvision_Example	Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 5, seq: 38, src: 0001, dst: 0006, transportPdu: 0xC2E3719D8130F5423D, netMic: 0xE6ECD1DC) encrypted using Primary Network Key (index: 0)
[... cut ...]
12-12 12:44:47.608 15007 15007 V OplusZoomWindowDecorViewHelper: setLastReportedMergedConfiguration mZoomDisplayHeight: 2400 getDecorView.144376167
12-12 12:44:47.621 15007 15007 I Quality : Skipped: false 3 cost 36.75503 refreshRate 0 processName no.nordicsemi.android.nrfmeshprovisioner
12-12 12:44:47.624 15007 15065 D SurfaceComposerClient: VRR [FRTC] client handle [bufferId:18446744073709551615 framenumber:0] [ffffffff, ffffffff]

I didn't go any further.

### Conclusion
Is the above behavior what you expect?

philips77, 2022-12-12:
Regarding

> GenericOnOffServer get of a node with IV 3 -> FAIL (timeout)

How many nodes do you have in your network? This looks like the Proxy Node or some relays have not been updated with the IV index. The ivi is set to 1, so seems like the IV index is used correctly. If that worked for IV Index 1 and 2, in my opinion should also for 3... Do you see logs from all nodes that they have received and applied the IV index update to 3?

Fabbros, 2022-12-14:
There are 3 nodes: A, B, and C:
- Node A is the one I used to trigger the IV update
- Node B is the one I tested with the GenericOnOff get/set
- Node C.. I only provisioned it to increase the number of devices in the same mesh :)

> If that worked for IV Index 1 and 2, in my opinion should also for 3...

I agree but at the moment the tests I've done say otherwise

> Do you see logs from all nodes that they have received and applied the IV index update to 3?

How can I be sure? From the app logs I only see the SNB from the proxy.

Fabbros, 2023-02-14:
Hi @philips77,
I ran a few tests again with the same 3 devices: with nRF Mesh Android (3.2.4) I was able to reach IV 4, normal op, and still able to get/set a GenericOnOff Server, nice!

Then I tried to step to IV 5 with app offline from the network, I checked that all 3 devices are on IV 5, normal op, and then connect to the network with app. 

```
02-14 12:42:05.259 18703 18703 D MeshManagerApi: Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 5, IV Update Active: false, Authentication Value: 0xA76ADC85DD599381}
02-14 12:42:05.259 18703 18703 D MeshManagerApi: Secure Network Beacon authenticated.
02-14 12:42:05.259 18703 18703 D MeshManagerApi: Last IV Index: 4
02-14 12:42:05.259 18703 18703 W MeshManagerApi: Discarding beacon IV Index: 5, IV Update Active: false, last 4, changed: 1h ago, test mode: true
```

philips77, 2023-02-22:
What was the last IV Update Active flag value when you put the phone offline? It can only update by 1 step, that is from IV Index 4 (flag active) to IV Index 5 (normal operation) unless you wait for 96h. If the flag was false, that it the last IV Index was 4 (normal operation), then it can't accept IV Index 5 with normal operation.

philips77, 2023-02-22:
It takes at least 96h to update by 1 step:
* IV Index X normal operation -> IV Index X update active
* IV Index X update active -> IV Index X+1 virtual operation

The test mode can shorten the transition by 96h only.

Fabbros, 2023-02-27:
I did the following:
1. put the app offline from the mesh, last IV with app online was IV 4 normal operation
2. stepped the mesh to IV 5 update active
3. stepped the mesh to IV 5 normal operation
4. put the app online on the same mesh with test flag enabled

> ```
> 02-14 12:42:05.259 18703 18703 D MeshManagerApi: Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 5, IV Update Active: false, Authentication Value: [...]

Putting that aside for a moment, I want to ask you a question: 
Hypothetically speaking, if I create a mesh with some devices, then wait a discrete amount of time (at least to make the mesh increase its IV to 2 for example), what happens if I want to do any edits to those devices? And what if I want to add a new device to mesh?

Halo3fanz, 2023-02-27:
> Hypothetically speaking, if I create a mesh with some devices, then wait a discrete amount of time to make the mesh increase its IV to 2 for example (like a year?), what happens if I go back to this mesh and I want to do any edits to those devices? And what if I want to add a new device to mesh?

I can answer that for you with experience; I have a mesh of 150 devices deployed at a client site. Last contact with the mesh from the provisioner was in early December. The mesh is quite high traffic; each device publishes a message every ten minutes. Just recently (read: today) I went back with the tablet I was using as a provisioner to integrate a new node. I saw all the symptoms you did: I can connect to the mesh but I cannot do "Get TTL" or any other commands. Provisioning a new device is successful but stops at getting composition of the device. I presume the device got updated to the correct IV index via a SNB packet but my tablet did not. It and the rest of the mesh are operating but I cannot configure or do anything from my tablet. IV index is now 2.

@philips77 I think there is an issue with the app not picking up the SNBs properly. I connected directly to a node that I configured as a SNB prior to this and sat there for an hour with IV test mode on. The tablet's messages are still being refused. I tried changing the provisioner's address as well but I'm having no luck.

Fabbros, 2023-02-27:
@Halo3fanz same here, I have more than one mesh with these symptoms, I also have a custom app built on an older nordic sdk that I want to upgrade to the latest version but, before doing it, wanna be sure that all these problem are fixed.

philips77, 2023-02-27:
Hi, are you also testing with iOS version of nRF Mesh? Do you have logs from Android version showing how the Snb is handled in connection?

Halo3fanz, 2023-02-28:
> Hi, are you also testing with iOS version of nRF Mesh? Do you have logs from Android version showing how the Snb is handled in connection?

Sure, its verbose but here you go:
[nrf_mesh_app_adb_log.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/10846970/nrf_mesh_app_adb_log.txt)

At line 1900-1944, you'll find an example of a command, likely the Get TTL command being sent out.
SNBs are being received at lines 712 and 1715 indicating IV index of 2 but the next lines show it discarding the beacon and remaining at IV index of 0.

Halo3fanz, 2023-02-28:
> @Halo3fanz same here, I have more than one mesh with these symptoms, I also have a custom app built on an older nordic sdk that I want to upgrade to the latest version but, before doing it, wanna be sure that all these problem are fixed.

Are you using nRF SDK for mesh or the new NCS?

Fabbros, 2023-03-06:
> Are you using nRF SDK for mesh or the new NCS?

The custom app uses:
- nRF SDK mesh v2.3.1 / Android
- nRF SDK mesh v2.2.3 / iOS 

I want to update to the latest version so I'm running some test before doing it.

What is NCS?

Halo3fanz, 2023-03-08:
@philips77 Please see my log above

Fabbros, 2023-03-13:
> > What is NCS?
> 
> The nRF Connect SDK is a new Zephyr-based SDK that Nordic is moving all its existing SDKs into. Regular BLE, Mesh and other RF technologies included. It's a radically different paradigm than their previous SDKs.
> 
> I assume you're updating to nRF SDK for Mesh v5.0,0? FYI, this SDK will not be updated further as it is in maintenance mode and Nordic recommends everyone to use NCS for new designs.

Actually, I'm interested in updating to nRF Mesh v3.2.4. Didn't know the existence of NCS, thank you!

Let's see what @philips77 has to say about your log.

Halo3fanz, 2023-03-31:
I just tested with an iOS device, and it worked! It worked without even enabling the IV test mode. I was able to provision and configure new devices with no issues. What does this mean? I much prefer the Android app's UI so I would still like this to be fixed.

IV Index is now 3 by the way.

If we're not getting replies from the dev team here, should we go to the devzone and find answers there?

philips77, 2023-03-31:
Hi, sorry for the long delay. I missed the notifications. @roshanrajaratnam will look into this soon. With DevZone I think it would still end up on our desks, so let's continue here.

Fabbros, 2023-03-31:
> Hi, sorry for the long delay. I missed the notifications. @roshanrajaratnam will look into this soon. With DevZone I think it would still end up on our desks, so let's continue here.

Yes, I already opened one there (see https://devzone.nordicsemi.com/f/nordic-q-a/92723/problem-with-iv-index-udpate) and they asked me to continue the discussion here.

> I just tested with an iOS device, and it worked! It worked without even enabling the IV test mode. I was able to provision and configure new devices with no issues. What does this mean? I much prefer the Android app's UI so I would still like this to be fixed.

Did the IV change while the provisioner was online? In this case, I don't have problem provisioning a new device too. 
If the IV change while the provisioner is offline the app will discard the beacon (tests donw with test mode enabled):

```
11:50:08.861656+0200	Discovering services...
11:50:11.821862+0200	Service found
11:50:11.821978+0200	Discovering characteristics...
11:50:13.328197+0200	Data In characteristic found
11:50:13.328547+0200	Data Out characteristic found
11:50:13.328711+0200	Enabling notifications...
11:50:15.073210+0200	Data Out notifications enabled
11:50:15.073273+0200	GATT Bearer open and ready
11:50:15.079289+0200	<- 0x010100CB8F04B8B03F507700000007A4974863FACF9815
11:50:15.079693+0200	Secure Network beacon (Network ID: CB8F04B8B03F5077, IV Index: 7 (normal operation), Key Refresh Flag: false) received (decrypted using key: Primary Network Key (index: 0))
11:50:15.080024+0200	Discarding beacon (IV Index: 7 (normal operation), last IV Index: 6 (normal operation), changed: 0h ago, test mode: true))
```

roshanrajaratnam, 2023-03-31:
@Fabbros so you mean that this works on iOS for you as well. I will check now!

Halo3fanz, 2023-03-31:
> Did the IV change while the provisioner was online?

The IV index had already rolled over to 3 when I introduced the iPad as a new provisioner using the same address as my Android device. In any case, I provisioned the new device while the iPad was connected to the network.

I'm not sure what would happen if it rolls over to 4 while I'm away; I'll have to come back in a month or so to check it out.

Fabbros, 2023-03-31:
@roshanrajaratnam everything is fine when the app is aligned with the mesh IV (and by that I mean that the IV changes while the app is connected to the mesh).

If the IV change while the app is offline then the app will discard the new SNB Beacon (even with IV test flag enabled) and keep using its value. This means that:
- if app IV = mesh IV - 1 -> message going out from the app will correctly be received by the peer but the vice-versa will be discarded by the app because unable to decrypt PDU
```
15:22:11.782672+0200	Secure Network beacon (Network ID: C4D2A5D77CD4505C, IV Index: 3 (normal operation), Key Refresh Flag: false) received (decrypted using key: Primary Network Key (index: 0))
15:22:11.787219+0200	Discarding beacon (IV Index: 3 (normal operation), last IV Index: 2 (normal operation), changed: 0h ago, test mode: true))
15:22:22.773559+0200	Sending GenericOnOffSet(tid: Optional(22), isOn: false, transitionTime: nil, delay: nil) from: Primary Element (0x0001), to: 0004
15:22:22.773597+0200	Sending Access PDU (opcode: 0x8202, parameters: 0x0016)
15:22:22.773832+0200	Sending Upper Transport PDU (encrypted data: 0x1291EF24, transMic: 0x9321FBC7) encrypted using key: App Key 1 (index: 0)
15:22:22.773874+0200	Sending Access Message (akf: 1, aid: 0x1A, szmic: 0, data: 0x1291EF249321FBC7)
15:22:22.773922+0200	Sending Network PDU (ivi: 0, nid: 0x7A, ctl: 0, ttl: 5, seq: 32, src: 0001, dst: 0004, transportPdu: 0x392269DB32F5A24B93, netMic: 0xF66C5D7C) encrypted using Primary Network Key (index: 0)
15:22:22.773979+0200	-> 0x007AA15FE0947DAF12BB392269DB32F5A24B93F66C5D7C
15:22:23.531164+0200	<- 0x00FA35D37870807A29D59045C1F98F06A9F4060021C032957B
15:22:23.532248+0200	Failed to decrypt PDU
15:22:24.281267+0200	<- 0x00FA691951704685748F7D001BB091986F5BEE02849960A204
15:22:24.282279+0200	Failed to decrypt PDU
15:22:24.780416+0200	<- 0x00FA93A77EEF13FD20F04213AB6159E9515A036AD580E1D796
15:22:24.781389+0200	Failed to decrypt PDU
15:22:25.531196+0200	<- 0x00FA946C133E52AF45466FAFD3587BF0902AF27E41C2AF5C79
15:22:25.532259+0200	Failed to decrypt PDU
15:22:26.283509+0200	<- 0x00FAB73C7CF7C85668E3540B5E472AEB05BB2F4CE79C751E14
```
- if app IV < mesh IV -1 -> nothing works..
```
15:27:12.800375+0200	Secure Network beacon (Network ID: C4D2A5D77CD4505C, IV Index: 4 (normal operation), Key Refresh Flag: false) received (decrypted using key: Primary Network Key (index: 0))
15:27:12.802642+0200	Discarding beacon (IV Index: 4 (normal operation), last IV Index: 2 (normal operation), changed: 0h ago, test mode: true))
15:27:21.764022+0200	Sending GenericOnOffSet(tid: Optional(24), isOn: true, transitionTime: nil, delay: nil) from: Primary Element (0x0001), to: 0004
15:27:21.764125+0200	Sending Access PDU (opcode: 0x8202, parameters: 0x0118)
15:27:21.767841+0200	Sending Upper Transport PDU (encrypted data: 0x37285196, transMic: 0x80271930) encrypted using key: App Key 1 (index: 0)
15:27:21.767949+0200	Sending Access Message (akf: 1, aid: 0x1A, szmic: 0, data: 0x3728519680271930)
15:27:21.768400+0200	Sending Network PDU (ivi: 0, nid: 0x7A, ctl: 0, ttl: 5, seq: 39, src: 0001, dst: 0004, transportPdu: 0xC5808110702E7EA6EE, netMic: 0x2E99F95E) encrypted using Primary Network Key (index: 0)
15:27:21.768462+0200	-> 0x007A4FED48B73C1FBF57C5808110702E7EA6EE2E99F95E
15:27:26.361915+0200	Resending Access PDU (opcode: 0x8202, parameters: 0x0118)
15:27:26.366965+0200	Sending Upper Transport PDU (encrypted data: 0x31BAAAFD, transMic: 0x4D4F4F7D) encrypted using key: App Key 1 (index: 0)
15:27:26.367115+0200	Sending Access Message (akf: 1, aid: 0x1A, szmic: 0, data: 0x31BAAAFD4D4F4F7D)
15:27:26.367959+0200	Sending Network PDU (ivi: 0, nid: 0x7A, ctl: 0, ttl: 5, seq: 40, src: 0001, dst: 0004, transportPdu: 0xE412D08F33AD78FB42, netMic: 0x42BC82FB) encrypted using Primary Network Key (index: 0)
15:27:26.368227+0200	-> 0x007A209C667DE43307E4E412D08F33AD78FB4242BC82FB
15:27:35.473979+0200	Resending Access PDU (opcode: 0x8202, parameters: 0x0118)
15:27:35.476783+0200	Sending Upper Transport PDU (encrypted data: 0x11A99AF4, transMic: 0xED16F9F4) encrypted using key: App Key 1 (index: 0)
15:27:35.476856+0200	Sending Access Message (akf: 1, aid: 0x1A, szmic: 0, data: 0x11A99AF4ED16F9F4)
15:27:35.477423+0200	Sending Network PDU (ivi: 0, nid: 0x7A, ctl: 0, ttl: 5, seq: 41, src: 0001, dst: 0004, transportPdu: 0x9BAD073FD82910149A, netMic: 0x2D3FCE7F) encrypted using Primary Network Key (index: 0)
15:27:35.477535+0200	-> 0x007AF0E3B92974913DB99BAD073FD82910149A2D3FCE7F
15:27:53.560262+0200	Resending Access PDU (opcode: 0x8202, parameters: 0x0118)
15:27:53.596734+0200	Sending Upper Transport PDU (encrypted data: 0xC459984A, transMic: 0x2D1896E3) encrypted using key: App Key 1 (index: 0)
15:27:53.596831+0200	Sending Access Message (akf: 1, aid: 0x1A, szmic: 0, data: 0xC459984A2D1896E3)
15:27:53.597279+0200	Sending Network PDU (ivi: 0, nid: 0x7A, ctl: 0, ttl: 5, seq: 42, src: 0001, dst: 0004, transportPdu: 0xCB0C972217F0E4EE22, netMic: 0xC025FBA7) encrypted using Primary Network Key (index: 0)
15:27:53.597330+0200	-> 0x007A56F60A5585E6162CCB0C972217F0E4EE22C025FBA7
15:28:01.843708+0200	Response to Access PDU (opcode: 0x8202, parameters: 0x0118) not received (timeout)
15:28:01.843988+0200	GenericOnOffSet(tid: Optional(24), isOn: true, transitionTime: nil, delay: nil) sent from: 0001, to: 0004 timed out
15:28:01.844259+0200	Cancelling messages with op code: 33282, sent from: 0001 to: 0004
```

roshanrajaratnam, 2023-03-31:
@Fabbros thanks for the details info, I'll get back to you on this.

philips77, 2023-04-01:
Hello, please remember, that IV Test Mode allows to skip 96h only. If, during the tests, you increment IV Index NY 2+ steps (a step means normal->active or active->normal) when the app is offline it will still reject the SNB until required time passed.

This is in regarding to https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/537#issuecomment-1446052770

philips77, 2023-04-01:
Also, in here
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/537#issuecomment-1491674412
You increased the IV Index using test mode by nie than one step when the app was offline. Such IV changes will be rejected. Test mode allowed to jump by one step in without waiting 96h. Just one step  all devices must be onboard, including the phone. When all nodes and phones accepted the new IV Index, you may jump to the next step.

philips77, 2023-04-01:
Test mode should only be used for testing. If you have mesh network in production and they're configured, wiring, changing IV Index when required, etc, they should respect all IV Index change time requirements. Then, the iphone can be offline for as long as 42 weeks and it should accept the SNB.

philips77, 2023-04-01:
Btw, there's a workaround for all IV Index issues.
When the phone rejects SNB do the following:
1. Export network configuration to json, cloud, wherever.
2. Erase application storage in Android settings. On iOS you may need to uninstall and reinstall the app, it there's some clever way to do that, as app data may be synced to iCloud and restored after reinstalling.
3. Import the same network.
4. Connect and try again.

The last IV Index and the time of the last chance are stored in the app storage. If no such exist, the app will accept any SNB it receive as the first one.

philips77, 2023-04-01:
> Then, the iphone can be offline for as long as 42 weeks and it should accept the SNB.

If you have your custom app based on nRF Mesh library you may disable this 42 week restriction. There an API for that. This is because what I wrote in the previous comment. Accepting any SNB is possible, so why making it difficult.

Halo3fanz, 2023-04-03:
> Btw, there's a workaround for all IV Index issues. When the phone rejects SNB do the following:
>
> 1. Export network configuration to json, cloud, wherever.
> 2. Erase application storage in Android [...]

Hi @philips77, I tried all of this and even took the extra step to uninstall the app and reboot my Android device but I still have no luck with communicating with my site network. I'm running Android 13 on a Samsung Tab S7.

Fabbros, 2023-04-03:
> If you have your custom app based on nRF Mesh library you may disable this 42 week restriction. There an API for that: [...]

So, just in case, I can use this as plan B, thank you!

BTW I don't understand one thing: If I have a configured mesh network that stepped to the next IV, I will be able to go back to that network and configure a new device without any issues right? The phone in this case will accept the SNB?

@philips77 please confirm that the following steps are the right ones to test this scenario:
1. create a new mesh network and configure 2/3 devices
2. disconnect the phone
3. manually force the next IV
4. wait for at least 96h
5. reconnect the phone and configure a new device

Also, what do you expect will happen in this one?
1. create a new mesh network and configure 2/3 devices
2. disconnect the phone
3. manually force the next IV
4. wait for at least 96h
5. manually force the next IV again so the phone will be behind 2 IVs
6. wait for at least 96h
7. reconnect the phone and configure a new device

philips77, 2023-04-03:
> If I have a configured mesh network that stepped to the next IV, I will be able to go back to that network and configure a new device without any issues right? The phone in this case will accept the SNB?

If the network was updating IV Index in the intervals specified by the spec, the phone should be able to accept SNB.

> please confirm that the following steps are the right ones to test this scenario:
> 1. create a new mesh network and configure 2/3 devices
> 2. disconnect the phone
> 3. manually force the next IV (normal -> active and then active -> normal)
> 4. wait for at least 96h
> 5. reconnect the phone and configure a new device

Each step in an IV Index should take at least 96h. A step is switching from "normal operation" to "update active" and another step is to switch back from "update active" to "normal operation". Full IV Index Update (normal->active->normal) takes at least 192h.
Setting "IV Test Mode" allows to decrease this time by 96h. It is intended for testing only. When in use, all devices, including the phone, should transition to the next step before starting new transition. 

> Also, what do you expect will happen in this one?
> 1. create a new mesh network and configure 2/3 devices
> 2. disconnect the phone
> 3. manually force the next IV
> 4. wait for at least 96h
> 5. manually [...]

If "manually force the next IV" is going 2 steps, than 96h in step 4 is not enough. 192h are required for the SNB to be accepted. 
In 5. You update the IV by 2, which is around 4 steps. 96h will not be enough.

Fabbros, 2023-04-03:
@philips77 now it's all clearer! Thank you! 

I'll be back after some tests.. :)

Halo3fanz, 2023-04-10:
Hi, @philips77 @roshanrajaratnam,

Any idea why the workaround is not resolving my issue? It seems like the SNBs are still getting rejected for some reason. I'm afraid the same thing might happen on my iOS device as well in the future.

roshanrajaratnam, 2023-04-11:
This seems to be fine.

Fabbros, 2023-04-12:
@philips77 with your help, I was successful able to do the following:
1. create a new mesh with 3 devices
2. disconnect the app
3. export the mesh configuration from nRF Mesh
4. roll IV to 4, normal operation
5. uninstall+install of nRF Mesh
6. import the mesh previously exported
7. connect the app

[image]

Urrà!

@roshanrajaratnam I still have problem with android, I did the very same step of iOS but SNB was discarded:

04-12 17:38:48.214 23054 23054 I BleManager: Services discovered
04-12 17:38:48.217 23054 23054 D BluetoothGatt: configureMTU() - device: E4:9A:F7:DD:7A:B1 mtu: 517
04-12 17:38:48.233 23054 23091 D BluetoothGatt: onConfigureMTU() - Device=E4:9A:F7:DD:7A:B1 mtu=69 status=0
04-12 17:38:48.239 23054 23054 I BleManager: MTU changed to: 69
04-12 17:38:48.241 23054 23054 D BluetoothGatt: setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
04-12 17:38:48.265 23054 23054 I BleManager: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
04-12 17:38:48.265 23054 23054 I BleManager: Notifications enabled
04-12 17:38:48.318 23054 23054 I BleManager: Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 01-01-00-D6-C8-E8-52-AF-44-75-85-00-00-00-03-2A-06-53-2E-DD-29-67-9D
04-12 17:38:48.332 23054 23054 D BluetoothAdapter: isLeEnabled(): ON
04-12 17:38:48.335 23054 23091 D BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
04-12 17:38:48.337 23054 23074 D BluetoothGatt: onConnectionUpdated() - Device=E4:9A:F7:DD:7A:B1 interval=198 latency=0 timeout=400 status=0
04-12 17:38:48.345 23054 23054 D MeshManagerApi: Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 3, IV Update Active: false, Authentication Value: 0x2A06532EDD29679D}
04-12 17:38:48.345 23054 23054 D MeshManagerApi: Secure Network Beacon authenticated.
04-12 17:38:48.345 23054 23054 D MeshManagerApi: Last IV Index: 0
04-12 17:38:48.345 23054 23054 W MeshManagerApi: Discarding beacon IV Index: 3, IV Update Active: false, last 0, changed: 0h ago, test mode: false

Halo3fanz, 2023-04-13:
> This seems to be fine.

But it's not working. There has to be something going wrong somewhere else. What could it be?

roshanrajaratnam, 2023-04-13:
Hey both, been a little bit busy with some other tasks. I'm looking in to this now as we speak.

roshanrajaratnam, 2023-04-14:
Hi @Fabbros and @Halo3fanz

First of all sorry for the LONG reply! I've just finished some tests regarding the IV Index update and the above said work around works as expected on the Android app as well. Please find the logs below for confirmation. 

The firmware used here is a mesh shell application that's used to test the IV Index update procedure. You can use this to bump the iv index value going through each update state until you reach the desired IV Index value.

To make it easier I have attached the shell app which is built for nRF 52840 board [zephyr_iv_test.zip](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/11230706/zephyr_iv_test.zip)

However if you would like to build it yourself here's the link to the shell app.
https://docs.zephyrproject.org/latest/connectivity/bluetooth/api/mesh/shell.html#application
In case you are building it yourself `CONFIG_BT_MESH_IV_UPDATE_TEST` is already enabled as shown here
https://github.com/zephyrproject-rtos/zephyr/blob/9e105d20376630170cdcd917e2ea32e5539cb3d4/tests/bluetooth/mesh_shell/prj.conf#L77

I've used the following steps to test the issue.

1. Provisioned a node and upon reconnection  the logs below shows that the IV Index is at 0
```
2023-04-14 09:48:47.174 D connect() - device: C9:24:BF:E7:C7:51, auto: false
2023-04-14 09:48:47.174 D registerApp()
2023-04-14 09:48:47.174 D registerApp() - UUID=4fc4c932-a1a1-46c7-8516-60586a6b8989
2023-04-14 09:48:47.176 D onClientRegistered() - status=0 clientIf=10
2023-04-14 09:48:47.641 D isLeEnabled(): ON
2023-04-14 09:48:48.100 D onClientConnectionState() - status=0 clientIf=10 device=C9:24:BF:E7:C7:51
2023-04-14 09:48:48.100 I Connected to C9:24:BF:E7:C7:51
2023-04-14 09:48:48.403 D discoverServices() - device: C9:24:BF:E7:C7:51
2023-04-14 09:48:48.544 D onConnectionUpdated() - Device=C9:24:BF:E7:C7:51 interval=6 latency=0 timeout=500 status=0
2023-04-14 09:48:48.545 I Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2023-04-14 09:48:48.705 D onSearchComplete() = Device=C9:24:BF:E7:C7:51 Status=0
2023-04-14 09:48:48.705 I Services discovered
2023-04-14 09:48:48.706 D configureMTU() - device: C9:24:BF:E7:C7:51 mtu: 517
2023-04-14 09:48:48.724 D onConfigureMTU() - Device=C9:24:BF:E7:C7:51 mtu=69 status=0
2023-04-14 09:48:48.724 I MTU changed to: 69
2023-04-14 09:48:48.724 D setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
2023-04-14 09:48:48.740 I Data written to descr. 00002902-0000-1000-8000-00805f9b34fb
2023-04-14 09:48:48.740 I Notifications enabled
2023-04-14 09:48:48.760 I Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 01-01-00-EF-B1-8C-7D-B9-47-A3-F0-00-00-00-00-47-50-FA-91-36-CA-FF-EB
2023-04-14 09:48:48.775 D Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 0, IV Update Active: false, Authentication Value: 0x4750FA9136CAFFEB}
2023-04-14 09:48:48.776 D Secure Network Beacon authenticated.
2023-04-14 09:48:48.776 D Last IV Index: 0
2023-04-14 09:48:48.776 I Setting IV Update Active to: false
2023-04-14 09:48:48.782 D isLeEnabled(): ON
2023-04-14 09:48:48.787 D onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2023-04-14 09:48:48.813 D onConnectionUpdated() - Device=C9:24:BF:E7:C7:51 interval=36 latency=0 timeout=500 status=0
2023-04-14 09:48:48.837 I Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
2023-04-14 09:48:48.909 D isLeEnabled(): ON
2023-04-14 09:48:50.741 D endAllActiveAnimators on 0x75c02fe3c0 (RippleDrawable) with handle 0x74d0320870
2023-04-14 09:48:52.819 D onConnectionUpdated() - Device=C9:24:BF:E7:C7:51 interval=36 latency=0 timeout=42 status=0
2023-04-14 09:48:52.819 I Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 420ms)
```

2. Disconnect from the device
3. Export the network.

>5. uninstall+install of nRF Mesh (to skip 96h * 2 * 4=768h)

4. @Fabbros @Halo3fanz  I have reset the Reset the network configuration in the app just to ensure that there is no old network information.
5. Bump IV index to 4 using the shell app

[image]

6. Import the mesh network
7. Reconnect to the node and you can see that once the secure network beacon is authenticated the IV index is updated to 4.

```
2023-04-14 09:54:11.742 D connect() - device: C9:24:BF:E7:C7:51, auto: false
2023-04-14 09:54:11.742 D registerApp()
2023-04-14 09:54:11.742 D registerApp() - UUID=b6a862f3-8ee5-4745-8ec5-00fb9285018a
2023-04-14 09:54:11.745 D onClientRegistered() - status=0 clientIf=10
2023-04-14 09:54:12.223 D isLeEnabled(): ON
2023-04-14 09:54:14.131 D onClientConnectionState() - status=0 clientIf=10 device=C9:24:BF:E7:C7:51
[... 83 lines cut ...]
2023-04-14 09:55:25.715 D onConnectionUpdated() - Device=C9:24:BF:E7:C7:51 interval=6 latency=0 timeout=500 status=0
2023-04-14 09:55:25.716 I Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2023-04-14 09:55:25.875 D onSearchComplete() = Device=C9:24:BF:E7:C7:51 Status=0
2023-04-14 09:55:25.876 I Services discovered
2023-04-14 09:55:25.877 D configureMTU() - device: C9:24:BF:E7:C7:51 mtu: 517
2023-04-14 09:55:25.894 D onConfigureMTU() - Device=C9:24:BF:E7:C7:51 mtu=69 status=0
2023-04-14 09:55:25.900 I MTU changed to: 69
2023-04-14 09:55:25.900 D setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
2023-04-14 09:55:25.924 I Data written to descr. 00002902-0000-1000-8000-00805f9b34fb
[... cut ...]
2023-04-14 09:55:26.016 I Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
2023-04-14 09:55:26.095 D isLeEnabled(): ON
2023-04-14 09:55:30.038 D onConnectionUpdated() - Device=C9:24:BF:E7:C7:51 interval=36 latency=0 timeout=42 status=0
2023-04-14 09:55:30.040 I Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 420ms)
```

Unfortunately I cannot see why it should fail in this case. Could you guys test against the test firmware to see if it behaves as expected?

Fabbros, 2023-04-20:
@roshanrajaratnam Unfortunately it seems I can't get things to work well.

I followed the same steps as yours but still 
```
04-20 15:04:25.652 22276 22276 D MeshManagerApi: Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 4, IV Update Active: false, Authentication Value: 0xB7D8DBC06FA146ED}
04-20 15:04:25.652 22276 22276 D MeshManagerApi: Secure Network Beacon authenticated.
04-20 15:04:25.652 22276 22276 D MeshManagerApi: Last IV Index: 0
04-20 15:04:25.652 22276 22276 W MeshManagerApi: Discarding beacon IV Index: 4, IV Update Active: false, last 0, changed: 0h ago, test mode: false
```
Even uninstalling + installing + reset + import seems to work, I also tried importing the mesh into a different Android phone but nothing.

Then I imported the same mesh on nRF Mesh iOS and everything worked fine :(

Fabbros, 2023-05-08:
[logcat.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/11420039/logcat.txt)

Hi @roshanrajaratnam, I attached the full trace as you requested. Feel free to ask any questions.

Fabbros, 2023-05-11:
[full trace 2.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/11449865/full.trace.2.txt)
@roshanrajaratnam I attached a second trace, this time I configured the mesh using iOS nRF Mesh, rolled IV to 5, exported the configuration to nRF Mesh Android. As you can see the beacon is still getting discarded :(

roshanrajaratnam, 2023-05-24:
@Fabbros sorry about the delayed reply, I will take a look at this case first thing tomorrow and get back to you with an update!

roshanrajaratnam, 2023-05-25:
@Fabbros looking at the logs, I see that the following log.
`05-11 10:04:35.495  4428  4428 W MeshManagerApi: Discarding beacon IV Index: 5, IV Update Active: true, last 0, changed: 0h ago, test mode: false`

Was the Android device connected to the network at some point when the IV index update began? for example when rolling from 0 - 4? If this was the case then the IV Index update will not succeed as it will not have completed the 96 hour time limit.

roshanrajaratnam, 2023-05-25:
Hi @Fabbros we see that there was indeed a bug on the app that's on production that explains the issue you are facing. This issue was fixed by the following PR [#524](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/524)

Could you pull the latest changes from the main branch and see if this fixes your issue. @Halo3fanz you could try the same if the issue persists.

Fabbros, 2023-05-25:
Hi @roshanrajaratnam that looks like it! I will try this one and update you as soon as possible.

Fabbros, 2023-05-26:
Early tests are positive, and now the SNB is not discarded anymore! :) 
I was also able to apply the patch to my custom app.

Halo3fanz, 2023-05-26:
> Hi @Fabbros we see that there was indeed a bug on the app that's on production that explains the issue you are facing. This issue was fixed by the following PR [...]

I'm facing issues building on Android Studio at the moment. Is 3.3 not making its way to the Play Store yet?

roshanrajaratnam, 2023-05-26:
Yup it's on its way!

roshanrajaratnam, 2023-05-30:
@Halo3fanz did you get to try this?

Halo3fanz, 2023-06-01:
> @Halo3fanz did you get to try this?

Yes! I updated the app, tested in the office and it worked fine. Then I brought the Android device to two of my customer sites and it worked as well. Great job @roshanrajaratnam!

closed 2023-06-01 by roshanrajaratnam