nordicsemi/Android-nRF-Mesh-Library #539: ConfigCompositionDataStatus message destroys subscribe information

Rawa, 2022-10-17:
---

**Describe the bug**
A ConfigCompositionDataStatus message will destroy subscribe information for a given node.

In our usecase we set up a mesh, with subscribe to groups etc. Then we use DFU directly through BLE, to ensure that the node is up to date we ask the node about it's composition data (to get updated VID). This seems to cause a all subscribes to be removed from the config even though they are in reality still there.

**To Reproduce**
Steps to reproduce the behavior:
1. Provision node
2. Add subscribe
3. Send message ConfigCompositionDataGet to update composition data.
4. Subscribe is now gone from the config.

**Expected behavior**
The subscribe should still be there if the element exists

**Platform details:**
 - Device: Pixel 6 Pro
 - OS: Android 12
 - Library Version (or Commit): 3.2.4

Rawa, 2022-10-17:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/2ef8d19b278ae311802924bb76fe50eb031423bd/mesh/src/main/java/no/nordicsemi/android/mesh/transport/ProvisionedMeshNode.java#L509

I believe this line could be the culprit, basically replacing the entries in the hashmap w/o taking into account what is already there

Rawa, 2022-10-17:
Tested it locally, confirmed that it is this line that will remove the subscribe information. It literally replaces all information already in the elements.

roshanrajaratnam, 2023-01-16:
I guess this relates to requesting composition data as the library was based on the assumption that the composition data request will be sent only once during configuration. I guess there seems to be more discussion around this.