nordicsemi/Android-nRF-Mesh-Library #542: How to control bluetooth mesh on other phones?

DM-Li-Lee, 2022-10-20:


philips77, 2022-10-20:
I transferred the issue here, as it's related to mesh, not BLE library.

If you want to control send data between 2 phones you need 2 other proxy nodes: one relaying packets to each phone: 
```mermaid
flowchart LR;
    A <-- GATT-->P1;
    P1 <-- ADV -->P2;
    P2 <-- GATT-->B;
```

philips77, 2022-10-20:
Mind, that the current version of the Android library does not support Server models automatically, not even Config Server. You can implement it on your own by handling selected message types and replying with Statuses. 

The iOS mesh library supports Config Server and Scene server models, and allows to define your own models, both server and client. We're working on bringing this functionality to Android as well, but this will come in 2023 (hopefully).

roshanrajaratnam, 2022-10-20:
Also if you just want to share the data between two phones you can export the mesh network and import it on phone B. Just create a new provisioner and you should be able to control without needing the config server support.

closed 2022-10-21 by philips77