nordicsemi/Android-nRF-Mesh-Library #543: What are the ProGaurd rules for the library

ZubairAkber, 2022-11-08:
---

**ProGaurd Rules for Mesh library**
I am using mesh library and have progaurd enable, can any one please share the rules for it, because i am getting errors i progaurd is enable in my project

**To Reproduce**
Steps to reproduce the behavior:
1. enable progaurd
2. build and run the app

philips77, 2022-12-13:
Hi,
This file: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/main/mesh/mesh-proguard-rules.pro should be added automatically to your Proguard settings thanks to this line: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/1fc9ce37c4b22effbc32c628a45dd33477d8013d/mesh/build.gradle#L43
Do you think something else is missing as well?

roshanrajaratnam, 2023-01-16:
Hi @ZubairAkber, Hope you managed to get this fixed. I will close this for now and please feel free to re-open if you need more help on this.

closed 2023-03-31 by roshanrajaratnam