nordicsemi/Android-nRF-Mesh-Library #544: Crash in nRF Mesh app version 3.2.4

PavelVPV, 2022-11-11:
---

**Describe the bug**
The app started suddenly crashing when trying to add new node. Neither force stop, nor data clear didn't work. Had to reboot the phone.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app
2. Press on `ADD NODE`
3. See error

**Expected behavior**
Usually, when pressing `ADD NODE`, you can see `CAN'T SEE YOUR NODE` text until devices start appearing. But when crash occurred, it was a blank frame for a second or two, then this text appeared and then crash happened.

**Platform details:**
 - Device: Realme GT Master Edition
 - OS: Android 12
 - Library Version: nRF Mesh app 3.2.4

**Logs / Screenshots**
```
11-11 16:04:10.355  1742  1742 D AndroidRuntime: Shutting down VM
--------- beginning of crash
11-11 16:04:10.357  1742  1742 E AndroidRuntime: FATAL EXCEPTION: main
11-11 16:04:10.357  1742  1742 E AndroidRuntime: Process: no.nordicsemi.android.nrfmeshprovisioner, PID: 1742
11-11 16:04:10.357  1742  1742 E AndroidRuntime: java.lang.IllegalArgumentException: scanner already started with given callback
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at no.nordicsemi.android.support.v18.scanner.f.c(Unknown Source:94)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at no.nordicsemi.android.support.v18.scanner.b.b(Unknown Source:30)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at no.nordicsemi.android.nrfmesh.viewmodels.b0.l(Unknown Source:73)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at no.nordicsemi.android.nrfmesh.ble.ScannerActivity.H1(Unknown Source:82)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at no.nordicsemi.android.nrfmesh.ble.ScannerActivity.s1(Unknown Source:0)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at no.nordicsemi.android.nrfmesh.ble.t.a(Unknown Source:4)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at androidx.lifecycle.LiveData.c(Unknown Source:29)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at androidx.lifecycle.LiveData.d(Unknown Source:44)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at androidx.lifecycle.LiveData.n(Unknown Source:14)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at androidx.lifecycle.LiveData$a.run(Unknown Source:18)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:938)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:99)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at android.os.Looper.loopOnce(Looper.java:233)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:344)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:8184)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
11-11 16:04:10.357  1742  1742 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1034)
```

philips77, 2022-11-19:
Thank you for your report. We will look into it.

roshanrajaratnam, 2023-01-16:
Hi @PavelVPV just got back from my paternity leave. I was unable to reproduce this issue on Android 13. I'll check with an Android 12 device and get back to you. The logs are quite clear as to what is happening.

closed 2023-03-31 by roshanrajaratnam