nordicsemi/Android-nRF-Mesh-Library #547: It is no longer possible to manually add nodes to the network

saty9, 2022-12-08:
Previously nodes could be added to the network by calling addNode on network with a provisionedMeshNode. But it is no longer possible to construct those since:

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/commit/6be990c99004903d04c6a5e292a551f70ee271ee#diff-91ebadcd566c87549cda1991db69903e973935b41a753e64bce3909251b474b9R103-R105

Which requires provisioning capabilities be set when constructing provisioned mesh nodes from unprovisioned ones. It is not possible to set the provisioning capabilites of an unprovisioned node or to construct a provisioning capabilities object however.

Is there some other mechanism for adding pre-existing nodes to a network now that I'm missing?

philips77, 2022-12-13:
Hello!

Thank you for bringing our attention to this aspect. We will look into that. Unfortunately, this may need to wait until next year due to holiday season.

roshanrajaratnam, 2023-01-16:
Hi @saty9, I just got back from my paternity leave. Let me take a look at this and get back to you.