nordicsemi/Android-nRF-Mesh-Library #552: Device connection failure

cjagdish, 2022-12-27:
---

**Describe the bug**
In our App, we are trying to connect automatically with ble mesh network. It's working fine with 70-80 devices. But when we add more then 100 devices then app is not able to connect with network.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a network of 120+ devices
2. When connecting with device, always connect with 1st device in scan list.
3. We are trying 3 attempts with same device if fails.

**Expected behavior**
Every time App is opened, network must be connected immediately in shorter time.

**Platform details:**
 - Device: Samsung A52
 - OS: Android 12
 - Library Version: 3.2.4

**Logs / Screenshots**
```
D/BluetoothGatt: connect() - device: 84:F7:03:16:39:9E, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=add37261-55ab-43ba-8f4b-3728ae000346
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
D/NrfMeshRepository: onDeviceConnecting
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=10 device=84:F7:03:16:39:9E
W/BleManager: Error: (0x85): GATT ERROR
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=10
D/BluetoothGatt: connect() - device: 84:F7:03:16:39:9E, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=864993da-f3f2-48a8-a513-b2334d0de24c
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
D/NrfMeshRepository: onDeviceConnecting
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=84:F7:03:16:39:9E
W/BleManager: Error: (0x85): GATT ERROR
W/BleManager: Connection attempt timed out
W/sole.commercia: Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (greylist, reflection, allowed)
D/BluetoothGatt: refresh() - device: 84:F7:03:16:39:9E
I/BleManager: Cache refreshed
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=9
D/method Name: connectDevice
D/event: {"connectDevice":"onDeviceDisconnected","eventName":"connectDevice"}
V/Capacitor/ReevaMeshPlugin: Notifying listeners for event connectDevice
D/NrfMeshRepository: onDeviceDisconnected
D/event: {"proxy":false,"eventName":"checkProxyConnectedOrNot"}
V/Capacitor/ReevaMeshPlugin: Notifying listeners for event checkProxyConnectedOrNot
D/onError_connecting: Error on connection state change
```

roshanrajaratnam, 2023-01-16:
Hi @cjagdish i've just got back from my paternity leave. The mesh library is functions independent of the ble transport and seems like a timeout (133) based on the logs. Are you still facing this issue?

cjagdish, 2023-01-19:
yes

roshanrajaratnam, 2023-03-31:
Unfortunately I don;t have a setup with 100 devices to replicate this issue. Have you tested if its reproduceable with other devices?

cjagdish, 2023-03-31:
yes we have tested other devices and facing same issue

anees17861, 2023-04-01:
Can the reason be simply too much traffic on the Bluetooth advertisement channels? If I'm not wrong, the connection also happens over advertising channels itself which is shared with mesh advertising.

Too much traffic could cause collision and hence add delay to the connection handshake

I'm assuming of course that these 120 devices are on and are pretty close.

roshanrajaratnam, 2023-05-25:
Could you try increasing the retry attempts? Currently the app retries 3 times if I remember correctly.

roshanrajaratnam, 2024-05-10:
Closing due to inactivity.

closed 2024-05-10 by roshanrajaratnam