nordicsemi/Android-nRF-Mesh-Library #553: Netkey command not following mesh specifications

Shane555, 2023-01-16:
---

**Describe the bug**
When i use the nRF mesh app to delete a netkey from my ble mesh mcu, the command config netkey delete received contains a net index of 0xffff instead of the net index that i wish to delete. The mesh specifications says 
[image]

**Expected behavior**
When i delete netkeys using the android app, for example, if i wish to delete net index 0x0002 from the device of interest, i will expect the delete netkeys command to send a message containing net index 0x0002.

**Platform details:**
 -  nRF mesh app Device: [Pixel 4A]
 - OS: [Android 13]
 - Version 3.2.4
- device of interest: ble mesh module

roshanrajaratnam, 2023-01-16:
Hi Shane, thanks for reporting this issue. I will take a look at this and get back to you!

Shane555, 2023-01-30:
@roshanrajaratnam hi, any news on this?

roshanrajaratnam, 2023-03-31:
@Shane555 this seems to be working fine on our end. I am able to add and remove the said key index. Note that the key indexes are 12-bits long and is packed to two octets. Check table 4.3.1.1.

closed 2023-03-31 by roshanrajaratnam