nordicsemi/Android-nRF-Mesh-Library #558: Linking sent message to received acknowledgement

nikolayP4, 2023-05-22:
Hi,
Is it possible in the current version of the library to link sent mesh message and the acknowledgement for it?
And if possible, could you explain how to do it as I didn't found any information on that.
Thank you in advance!

roshanrajaratnam, 2023-05-25:
Hi this is handled by the library itself. The library holds a reference tot he last sent message. Check https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/a5cd85a7b771e1f6ba08fba707bd4a53cfa7adf0/mesh/src/main/java/no/nordicsemi/android/mesh/transport/DefaultNoOperationMessageState.java#L93 What do you plan to achieve?

nikolayP4, 2023-05-26:
I would like to log the messages to a backend. For example if I send an acknowledged message, I would like to log when it was sent and when I receive the response for this specific message, I want to log when I received the acknowledgement. In the log timestamps are included so it is important to link exact message sent with acknowledgement specifically for this message.

roshanrajaratnam, 2023-06-01:
Feel free to close the issue, if you managed to solve this.

nikolayP4, 2023-06-02:
I took a look into that class, but it is still not clear to me how can I (as a user of the library) compare acknowledgement that I have received to a message that I have sent. 
Currently I have the following problem. I store the messages that I have sent and when I receive acknowledgement, I compare the sources/destinations and the type of the message. It is working for some time, until one of the messages is lost. E.g. 
I send Message1 (GenericOnOffSet) to the node and store it. Message1 is lost and I don't receive acknowledgement for it. I send Message2 (GenericOnOffSet) the same type of message to the same node and I receive acknowledgement for it. The app then checks all messages that are sent and it stumbles upon stored Message1, the source and the type of message is accepted and the system thinks that the acknowledgement received is for Message1, however it is for Message2.

roshanrajaratnam, 2026-08-27:
The library handles segmented messages itself.  Closing this due to inactivity.

closed 2026-08-27 by roshanrajaratnam