nordicsemi/Android-nRF-Mesh-Library #562: When will the flutter version be released?

liuhk8, 2023-07-12:
When will the flutter version be released?

roshanrajaratnam, 2023-07-12:
Hi we do not have any flutter implementation for mesh. We are aware of some customers who have implemented a wrapper around our mesh implementations.

- https://pub.dev/packages/nordic_nrf_mesh

- https://github.com/OZEO-DOOZ/nrf_mesh_plugin

Hope this helps.

closed 2023-07-12 by roshanrajaratnam

liuhk8, 2023-07-13:
> Hi we do not have any flutter implementation for mesh. We are aware of some customers who have implemented a wrapper around our mesh implementations.
> 
> - https://pub.dev/packages/nordic_nrf_mesh
> 
> - https://github.com/OZEO-DOOZ/nrf_mesh_plugin
> 
> Hope this helps.

Thank you for your reply. In fact, I have tried both of these. It seems that custom models or vendor models are not supported.

roshanrajaratnam, 2023-07-13:
Hey please do get in touch with them and see if they could support your case. They have asked us to divert any flutter mesh related requests.