nordicsemi/Android-nRF-Mesh-Library #566: Key Refresh Procedure

BlackWolf4k, 2023-08-29:
Hi @roshanrajaratnam,
I'm trying to implement the **remotion of a node** from the network which as I can see by now it is **not** implemented in this source code yet, but I have two questions:
In the first step of this procedure ( *Specification v1.0.1 section 3.10.4.1* ) should I connect to each single node in the network and send to each the NetKey Update ( *Specification v1.0.1 section 4.3.2.32* ) and AppKey Update ( *Specification v1.0.1 section 4.3.2.38* ) messages?
And as second question: In the Phase 2 of the procedure ( *Specification v1.0.1 section 3.10.4.2* ) as I can't send a Secure Network Beacon because the device I'm using to configure the nodes is not itself part of the network, is it enough to send the Config Key Refresh Phase Set to just one node to update the whole network?

closed 2023-08-30 by roshanrajaratnam

BlackWolf4k, 2023-08-30:
Hi again @roshanrajaratnam,
I have been trying to implement this procedure following both the *Specification v1.0.1 section 3.10.4* and PR #381 referred by you and I have the following issue:
As the **first** message of the whole procedure I sent the `NetKey Update` where as parameters I sent:
As NetKey Index: 0;
As NetKey of course a different one from the previous;
After the message has been sent and the `NetKey Status` message has been received the board stops answering to any other message.
Am I right sending the same NetKey Index of the old key? And am I right sending as following message the `ConfigKeyRefreshPhaseSet` ( encoded with the new NetKey )?
( I have not implemented the usage of Proxy for this step yet )

roshanrajaratnam, 2023-08-31:
Hi, according to the spec, you should be sending the original KeyIndex (KeyIndex never changes) and the new key. If I remember correctly you should send the `ConfigKeyRefreshPhaseSet` encrypted with the old key.  According to step 10 in the PR, once the phase is set, it can receive messages with both old net key and the new net key. Before changing the phase you are still distributing the keys phase which means that your new net key is not ready for use yet, hence you must send the messages with the old key.

BlackWolf4k, 2023-09-06:
Hi again @roshanrajaratnam,
I have implemented the *Key Refresh Procedure* almost completely thanks to your suggestions but I still have one issue:
When sending the `ConfigKeyRefreshPhaseSet`( **encoded with the old key** ) message to go from Phase 1 to Phase 2 or from Phase 1 to Phase 3, I receive a response ( `ConfigKeyRefreshPhaseStatus` ) message which:
Has the Nid that is derived from the **New NetKey** but the content of the message is encoded with the keys derived from the **Old NetKey**.
But from what I have understood from the *Specification v1.0.1* during both Phase 2 and Phase 3 the node should be responding using the keys derived from the New NetKey.
Also the content of the response message is what I am expecting:
Phase 2: `80 17 00 00 00 02`
Phase 3: `80 17 00 00 00 00`
And the **Secure Network Beacon** also changes in the expected one ( the one that refers to the New NetKey )
Would you have any idea on why this happens?

roshanrajaratnam, 2023-09-07:
According to Spec during phase 2 the nodes are notified to use the new key. However you can still receive messages encrypted with both the **new** key and the **old** key. After switching to Phase 3 the **old** keys are revoked and the nodes will only transmit using the **new** key. Also the derived NID will refer to the same network key that was used to encrypt the message.