nordicsemi/Android-nRF-Mesh-Library #567: Can't set ConfigHeartbeatPublication countLog to 0xFF

QuentinBreant, 2023-10-11:
---

**Describe the bug**
Impossible to set the `ConfigHeartbeatPublicationSet countLog` to 0xFF. Parameter in constructor is a `byte` so the maximum I can pass is 127. I passed -1 to have 0xFF value equivalent to 255 but the `ConfigHeartbeatPublicationSet` constructor call the method `MeshParserUtils.isValidHeartbeatCountLog(final int countLog)` to verify the value. Problem is the byte passed in parameter is automatically casted as an Int and my 0xFF become 0xFFFFFFFF and method throws an `IllegalArgumentException`

**To Reproduce**
```
ConfigHeartbeatPublicationSet(
        node.unicast, -1, 2, 5,
        Features(
            Features.ENABLED,
            Features.ENABLED,
            Features.ENABLED,
            Features.ENABLED
        ), 0
    ) 

```
**Expected behavior**
I can pass the maximum value allowed for countLog.

**Platform details:**
 - Device: Samsung Galaxy A10 (SM-A105G/DS)
 - OS: Android 10
 - Library Version (or Commit): 3.3.0

roshanrajaratnam, 2023-10-16:
@QuentinBreant  try setting 0xFF directly and this should work for you I believe.

closed 2023-10-16 by roshanrajaratnam

QuentinBreant, 2023-10-17:
@roshanrajaratnam  I've got "The integer literal does not conform to the expected type Byte", 0xFF convert value to 255 which is bigger than the maximum Java byte type value.