nordicsemi/Android-nRF-Mesh-Library #568: Change abstract ApplicationMessage to public

anhhh11, 2023-10-12:
no\nordicsemi\android\mesh\transport\ApplicationMessage.java
abstract class ApplicationMessage extends MeshMessage

Currently, the app I am working on relies on the functionality provided by the ApplicationMessage class. However, the class is currently inaccessible or internal, which prevents me from customizing its content to suit my application's specific requirements. 
I believe making the ApplicationMessage class public would greatly benefit not only my application but also other developers who may encounter similar requirements. By allowing customization of this class, it would enable us to tailor our applications more effectively and efficiently.

roshanrajaratnam, 2023-10-13:
Hi ok, could you elaborate what you are trying to achieve?

anhhh11, 2023-10-13:
Yes, I want to create custom SIG model,.i.e send custom command (other
opcode + data format), some logic currently cast type to ApplicationMessage
but it cannot be extend.

roshanrajaratnam, 2023-10-13:
You can use vendor model message class to send custom messages.

anhhh11, 2023-10-13:
I find VendorModelMessageUnacked but `assembleMessageParameters` but it is final. Can you provide some class name or sample?

roshanrajaratnam, 2023-10-13:
Take a look at the following snippet, this should hopefully help you.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/1e71dabe7cd0a1047db4b662d2af41fb5e9e3584/app/src/main/java/no/nordicsemi/android/nrfmesh/node/VendorModelActivity.java#L223-L240

anhhh11, 2023-10-13:
Thank you very much! exactly what I need.

anhhh11, 2023-10-13:
Library already having VendorModelMessageAcked, VendorModelMessageUnacked for custom (vendor) message

closed 2023-10-13 by anhhh11