nordicsemi/Android-nRF-Mesh-Library #572: ETA on the release of the new kotlin version

Fabbros, 2023-10-30:
Hello!
I saw that a new version of the iOS SDK (with support for Bluetooth mesh protocol v1.1, mesh model v1.1, and more) was released about two weeks ago. What is the ETA for the new library written in Kotlin for Android?

roshanrajaratnam, 2023-10-31:
Hi Fabbros, we are currently working on a Kotlin mesh library that will include the mesh 1.1 features and the Java library will go in to maintenance mode. The plan is to release an early version of the kotlin library towards the end of the year.

WeiSunny, 2024-01-12:
> Hi Fabbros, we are currently working on a Kotlin mesh library that will include the mesh 1.1 features and the Java library will go in to maintenance mode. The plan is to release an early version of the Kotlin library towards the end of the year.

Hi roshanrajaratnam, has the new version of the Android SDK (supporting mesh 1.1 features) been released? 
Thanks

roshanrajaratnam, 2024-01-16:
We are still working on the internals. we will keep you posted about the release.

stevebutler11, 2024-07-17:
Hi @roshanrajaratnam, just wondering if there are any updates on the kotlin mesh library and its release?

roshanrajaratnam, 2024-07-17:
Hi, we are still working on this at the moment. The networking stack and the same app is still under development/testing and is not ready for production yet. I can't promise any dates as we have been working on some other projects at the same time as well. We are still a few months away from getting something out.

Bosemani, 2025-01-20:
Hai @roshanrajaratnam 
 Do you have Any update on kotlin mesh library?

roshanrajaratnam, 2025-01-20:
Hi, we are still working on this. We do have a stack in place that's currently under testing. We are not far off from releasing an early experimental version.

lecnierfreeman, 2025-04-30:
@roshanrajaratnam Any updates on when the Kotlin experimental version will be released?

connyhald, 2025-12-19:
I think you can close here. Kotlin version is available now: https://github.com/NordicSemiconductor/Kotlin-Mesh-Library/

closed 2026-01-07 by roshanrajaratnam