nordicsemi/Android-nRF-Mesh-Library #576: Create a group just after the library initialization causes the group to be created twice in database

JulesDommartin, 2023-11-23:
---

**Describe the bug**
When I create a group just after the library initialization, the group is created twice in database.

I tried with a bunch of different devices, and it always seems to create it twice anyway. 

**To Reproduce**
Steps to reproduce the behavior:
1. You create an empty project, and you import the library.
2. In the "onNetworkLoaded" callback, you create a group, like so:
[image]

3. You close the app
4. You reopen the app, group is created twice.

I created a repo with the code so you can also test it: [https://github.com/JulesDommartin/nrf-test-app](https://github.com/JulesDommartin/nrf-test-app)

**Expected behavior**
The group should be only created once.

**Platform details:**
 - Device: Samsung A13
 - OS: Android 13
 - Library Version : 3.3.2

**Logs / Screenshots**
I used the debugger to see it, so at step 2, when I just created my group, all seems fine:
[image]
[image]

But after step 3, when I reopen the app, and I stop at the same breakpoint than before, I have this: 
[image]

As you can see, the same group is created twice, with the same address, which is a big problem. 

To avoid that, just after creating the group, you can export and import the network, which forces the database to refresh the data with the current network loaded in RAM. 
`meshManagerApi.importMeshNetworkJson(meshManagerApi.exportMeshNetwork());`
Also, I tried to add a slight delay of 500ms between the "onNetworkLoaded" and the "createDefaultGroup" function, and it works on powerful smartphones, but not on older ones, as CrossCall M4.

I hope that all information will help to understand the problem, and eventually to fix it.

roshanrajaratnam, 2023-11-27:
@JulesDommartin  thanks for  reporting this issue and the detailed description. This is something we have not seen before. I will try to take a look at this.

doctorbebo, 2024-02-29:
has there been any updates on fix for this bug? I am running into the same issue.