nordicsemi/Android-nRF-Mesh-Library #583: Whats the advantage of generating a mesh network without an app key?

FluffyBunniesTasteTheBest, 2024-06-12:
Since version 3.3.3 `generateMeshNetwork()` (MeshManagerApi.java) no longer calls `generateAppKeys()`. Is there any advantage gained by it? Or any other reasons for it? 

A mesh network without at least a single app key is not very useful. Beside that, upgrading to 3.3.3 or 3.3.4 will break any existing apps that don't generate app keys themself, which makes this a rather risky change. One that might be worth reverting...

Question is, was this change made on purpose, and is it about to stay?

roshanrajaratnam, 2024-06-13:
Hi yes this was a change made on purpose it will stay. This is merely an application feature and should not be a part of the library. Users can always add app keys manually )

closed 2024-06-13 by roshanrajaratnam

FluffyBunniesTasteTheBest, 2024-06-13:
Considering that this is a breaking change it should at least be mentioned at the release notes... Anyway, thanks for the clarification.