nordicsemi/Android-nRF-Mesh-Library #584: Need help: Device Provisioning and Configuration Issue in NordicMesh Library for Android

ahmadshoh-orc, 2024-06-26:
I am currently implementing device provisioning and configuration using the NordicMesh library for Android. The implementation is done in Flutter with native Kotlin as a plugin. The provisioning process works correctly, but the device fails to reconnect and configure after provisioning.

#### Problem Statement
After successful provisioning, the device does not reconnect automatically, and the configuration process does not start. The `provisionCallback` is called, indicating provisioning success, but the subsequent connection attempt fails to trigger the configuration.

#### Code
Here is the relevant code for provisioning and configuration:

```kotlin
fun provision(address: String, callback: (provisionedNode: ProvisionedNode) -> Unit) {
    provisioningService.provisionCallback = { selectedPeripheral, unicastAddress ->
        Log.d("QBleMesh", "Provisioning device with address: $address")

        bleManager.connect(address) {
            Log.d("QBleMesh", "Connected to device with address: $address")
            startConfigurationProcess(selectedPeripheral, unicastAddress, callback)
        }
    }

    val unprovisionedScanResult = unprovisionedScanResults.firstOrNull { it.device.address == address }
    unprovisionedScanResult?.let { result ->
        bleManager.connect(address) {
            Log.d("QBleMesh", "Connection done was called.")
            provisioningService.startProvisioningProcess(result)
        }
    }
}

private fun startConfigurationProcess(
    selectedPeripheral: ScanResult,
    unicastAddress: Int,
    callback: (provisionedNode: ProvisionedNode) -> Unit
) {
    val name = selectedPeripheral.device.name ?: "unknown device"
    val address = selectedPeripheral.device.address

    unprovisionedScanResults.removeIf { it.device.address == address }

    val configurationService = ConfigurationService(meshManagerApi)
    configurationService.configurationComplete = { provisionedNode ->
        callback(provisionedNode)
    }

    configurationService.configure(unicastAddress, name, address)
}
```

#### Issue Details
The `provisionCallback` is triggered, and the connection is attempted but fails to proceed to the configuration stage. Below is the `BleMeshManager` class and the output logs during the provisioning process:

```kotlin
// BleMeshManager class implementation
class BleMeshManager(appContext: Context) : BleManager(appContext) {

    companion object {
        private const val MTU_SIZE_MAX: Int = 517
        private val MESH_PROVISIONING_DATA_IN: UUID =
            UUID.fromString("00002ADB-0000-1000-8000-00805F9B34FB")
        private val MESH_PROVISIONING_DATA_OUT: UUID =
            UUID.fromString("00002ADC-0000-1000-8000-00805F9B34FB")
        private val MESH_PROXY_DATA_IN = UUID.fromString("00002ADD-0000-1000-8000-00805F9B34FB")
        private val MESH_PROXY_DATA_OUT = UUID.fromString("00002ADE-0000-1000-8000-00805F9B34FB")
    }

    private var gatt: BluetoothGatt? = null
    private var meshProvisioningDataInCharacteristic: BluetoothGattCharacteristic? = null
    private var meshProvisioningDataOutCharacteristic: BluetoothGattCharacteristic? = null
    private var meshProxyDataInCharacteristic: BluetoothGattCharacteristic? = null
    private var meshProxyDataOutCharacteristic: BluetoothGattCharacteristic? = null

[... 243 lines cut ...]
D/BluetoothGatt( 6676): onServiceChanged() - Device=XX:XX:XX:XX:A5:7E
D/BluetoothAdapter( 6676): semIsBleEnabled(): ON
D/BluetoothAdapter( 6676): getBleEnabledArray(): ON
D/BluetoothAdapter( 6676): getBleEnabledArray(): ON
D/BluetoothAdapter( 6676): semIsBleEnabled(): ON
D/BluetoothAdapter( 6676): getBleEnabledArray(): ON
D/BluetoothLeScanner( 6676): Start Scan with callback
D/BluetoothLeScanner( 6676): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/BleMeshManager( 6676): provisioning is complete. unicast address is -> 3
D/BleMeshManager( 6676): we did it.
D/QBleMesh( 6676): Provisioning device with address: F0:F5:BD:85:A5:7E
D/BluetoothGatt( 6676): cancelOpen() - device: F0:F5:BD:85:A5:7E
D/BluetoothGatt( 6676): discoverServices() - device: F0:F5:BD:85:A5:7E
D/BluetoothGatt( 6676): onClientConnectionState() - status=0 clientIf=6 device=XX:XX:XX:XX:A5:7E
D/BluetoothGatt( 6676): close()
D/BluetoothGatt( 6676): unregisterApp() - mClientIf=6
```

#### Additional Context
- The `BleMeshManager` class handles the BLE Mesh management, including connection, provisioning, and data transfer for mesh devices.
- The logs indicate that the provisioning messages are being sent and received correctly.
- The configuration process does not start after the device reconnects post-provisioning

#### Request for Assistance
Any guidance on how to ensure the device reconnects and starts the configuration process automatically after provisioning would be greatly appreciated.

ahmadshoh-orc, 2024-06-26:
I searched a little further and it seems to me that I should not disconnect manually and i removed disconnect code from BleMeshManager. Now new connect function looks like this:

```
  fun connect(address: String, successCallback: SuccessCallback) {
    val bluetoothManager: BluetoothManager? = context.getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager?
    val device: BluetoothDevice? = bluetoothManager?.adapter?.getRemoteDevice(address)
    device?.run {
      connect(device).done(successCallback).fail { device: BluetoothDevice, status: Int ->
        gatt = null
        Log.d("BleMeshManager", "${device.name} failed with status -> $status")
      }.retry(3, 500).enqueue()
    }
  }
```

And below are new logs after provision. now it makes new connection and calls startConfiguration code, which should get CompositionData and start configuration. but now i see in logs, that MeshMessage was processed, but response not received:

```
D/BleMeshManager(12744): provisioning is complete. unicast address is -> 2
D/BleMeshManager(12744): we did it.
D/QBleMesh(12744): Provisioning device with address: F0:F5:BD:85:A5:7E
D/BluetoothGatt(12744): discoverServices() - device: F0:F5:BD:85:A5:7E
D/QBleMesh(12744): Connected to device with address: F0:F5:BD:85:A5:7E
V/MeshTransport(12744): Src address: 0001
V/MeshTransport(12744): Dst address: 0002
V/MeshTransport(12744): Key: B967907B068A4EBE422CB3A0BD3B86DC
V/MeshTransport(12744): akf: 0
V/MeshTransport(12744): aid: 0
V/MeshTransport(12744): aszmic: 0
V/MeshTransport(12744): Sequence number: 2
V/MeshTransport(12744): Access message opcode: 8008
V/MeshTransport(12744): Access message parameters: 00
V/AccessLayer(12744): Created Access PDU 800800
V/UpperTransportLayer(12744): Device nonce: 02000000020001000200000000
V/UpperTransportLayer(12744): Encrypted upper transport pdu: D3CB7E5F88EACF
V/LowerTransportLayer(12744): Unsegmented Lower transport access PDU 00D3CB7E5F88EACF
V/NetworkLayer(12744): Encryption key: A0710267FBC3D744FC60623B2E5DF416
V/NetworkLayer(12744): Privacy key: 6474BA4C7769978C5190AAAC575F9086
V/NetworkLayer(12744): Sequence Number: 000002
V/NetworkLayer(12744): Encrypted Network payload: 3678319FF25B8F41DE54A802753F
D/BluetoothAdapter(12744): getBleEnabledArray(): ON
D/BluetoothAdapter(12744): getBleEnabledArray(): ON
D/BluetoothAdapter(12744): semIsBleEnabled(): ON
D/BluetoothAdapter(12744): getBleEnabledArray(): ON
D/BluetoothAdapter(12744): getBleEnabledArray(): ON
D/BluetoothAdapter(12744): semIsBleEnabled(): ON
D/BluetoothAdapter(12744): getBleEnabledArray(): ON
D/BluetoothLeScanner(12744): Start Scan with callback
D/BluetoothLeScanner(12744): onScannerRegistered() - status=0 scannerId=9 mScannerId=0
D/ConfigurationService(12744): Mesh message processed -> dst: 2, meshMessage: 32776
D/BluetoothGatt(12744): onConnectionUpdated() - Device=XX:XX:XX:XX:A5:7E interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(12744): onSearchComplete() = Device=XX:XX:XX:XX:A5:7E Status=0
D/BluetoothGatt(12744): configureMTU() - device: F0:F5:BD:85:A5:7E mtu: 517
D/BluetoothGatt(12744): onConfigureMTU() - Device=XX:XX:XX:XX:A5:7E mtu=23 status=0
D/BluetoothGatt(12744): setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
D/BluetoothGatt(12744): setCharacteristicNotification() - uuid: 00002adc-0000-1000-8000-00805f9b34fb enable: true
D/BluetoothGatt(12744): onConnectionUpdated() - Device=XX:XX:XX:XX:A5:7E interval=24 latency=0 timeout=500 status=0
```

```
open class ConfigurationServiceMeshStatusCallback : MeshStatusCallbacks {
  protected var compositionDataGetComplete: () -> Unit = {}

  override fun onTransactionFailed(dst: Int, hasIncompleteTimerExpired: Boolean) {
    Log.d("ConfigurationService", "Transaction failed -> dst: $dst, hasIncompleteTimerExpired: $hasIncompleteTimerExpired")
  }

  override fun onUnknownPduReceived(src: Int, accessPayload: ByteArray?) {
    Log.d("ConfigurationService", "Unknown PDU received -> src: $src, accessPayload: ${accessPayload?.contentToString()}")
  }

  override fun onBlockAcknowledgementProcessed(dst: Int, message: ControlMessage) {
    Log.d("ConfigurationService", "Block acknowledgement processed -> dst: $dst, message: $message")
  }

  override fun onBlockAcknowledgementReceived(src: Int, message: ControlMessage) {
    Log.d("ConfigurationService", "Block acknowledgement received -> src: $src, message: $message")
  }

  override fun onMeshMessageProcessed(dst: Int, meshMessage: MeshMessage) {
    Log.d("ConfigurationService", "Mesh message processed -> dst: $dst, meshMessage: ${meshMessage.opCode}")
  }

  override fun onMeshMessageReceived(src: Int, meshMessage: MeshMessage) {
    Log.d("ConfigurationService", "Mesh message received -> src: $src, meshMessage: ${meshMessage.opCode}")
  }

  override fun onMessageDecryptionFailed(meshLayer: String?, errorMessage: String?) {
    Log.d("ConfigurationService", "Message decryption failed -> meshLayer: $meshLayer, errorMessage: $errorMessage")
  }
}
```

philips77, 2024-06-28:
Hello,
We are quite busy with other projects at the moment. We will come back to you as soon as possible. Sorry!

ahmadshoh-orc, 2024-07-02:
Hello,

I wanted to share my latest observations and compile all my problems and questions together briefly.

1. My issue closely resembles this issue: #570.
2. Provisioning works fine.
3. After provisioning, I am trying to reconnect to the node to fetch new service data.

**QUESTION:** Is it correct just call "connect" function from Ble-Library for reconnection?

```kotlin
bleManager.connect(address) {
    Log.d("QBleMesh", "Connected to device with address: $address")
    handler.postDelayed({
        Log.d("QBleMesh", "Starting configuration process.")
        startConfigurationProcess(selectedPeripheral, unicastAddress, callback)
    }, 5000)
}
```

**Details of my attempts:**

3.1. I attempted to call the connect function directly. It reconnects to the node, fetches new services and characteristics, and I can send `ConfigAppKeyAdd` and `ConfigCompositionDataGet`. Mesh messages are processed, but the problem is that no response messages are received, as seen in the logs.

3.2. I tried manually disconnecting from the provisioned node first and then reconnecting. It reconnects, but the "done" callback is not called, so I cannot start the configuration process.

```kotlin
bleManager.disconnect().enqueue()

bleManager.connect(address) {
  // Configuration
}
```

---

I have no ideas at all about what the problem might be and how to solve it. I would appreciate any insights or suggestions on how to resolve it. Thank you!

roshanrajaratnam, 2024-07-03:
Hello,

When it comes to connecting to a mesh node we don't use the concept of connecting using a address. Oncea node is provisioned, it belongs to a mesh network and all devices in the network advertises with a certain network id derived using the primary network key that was added during provisioning. When connecting to a provisioned mesh node or the proxy node you have to always filter the devices using their network ID. May be you have this already in place, however an example of how to implement could be found in the sample app.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/162a29e1b9b9de4c15f39f11c9034d51f486b1bd/app/src/main/java/no/nordicsemi/android/nrfmesh/viewmodels/ScannerRepository.java#L74-L99

> D/BluetoothGatt(13013): setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
> D/BluetoothGatt(13013): setCharacteristicNotification() - uuid: 00002adc-0000-1000-8000-00805f9b34fb enable: true
According to your logs I see the second characteristic seems to be wrong.

Following are the characteristics for a mesh proxy node with the mesh proxy service.
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/162a29e1b9b9de4c15f39f11c9034d51f486b1bd/app/src/main/java/no/nordicsemi/android/nrfmesh/ble/BleMeshManager.java#L66-L71

Let me know how this goes.

ahmadshoh-orc, 2024-07-04:
Hello Roshanrajaratman,

Thank you for your response. I implemented the reconnection method as you suggested. However, the logs indicate that the correct characteristics are initially set, but then overwritten by the provisioning characteristics. I am unsure why this occurs. I debugged each step, but I do not see any point in the code where the provisioning characteristics are being set after reconnection.

Here are the relevant log entries:

```
V/MeshManagerApi(18345): Received provisioning message: 0x0308
D/BleMeshManager(18345): provisioning is complete. unicast address is -> 2
D/BleMeshManager(18345): we did it.
D/BluetoothGatt(18345): onServiceChanged() - Device=XX:XX:XX:XX:08:92
D/BluetoothGatt(18345): cancelOpen() - device: 7C:DF:A1:DB:08:92
D/BluetoothGatt(18345): discoverServices() - device: 7C:DF:A1:DB:08:92
D/BluetoothGatt(18345): onClientConnectionState() - status=0 clientIf=6 device=XX:XX:XX:XX:08:92
D/BluetoothGatt(18345): close()
D/BluetoothGatt(18345): unregisterApp() - mClientIf=6
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothAdapter(18345): semIsBleEnabled(): ON
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothLeScanner(18345): Start Scan with callback
D/BluetoothLeScanner(18345): onScannerRegistered() - status=0 scannerId=9 mScannerId=0
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothLeScanner(18345): Stop Scan with callback
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothGatt(18345): connect() - device: 7C:DF:A1:DB:08:92, auto: false
D/BluetoothGatt(18345): registerApp()
D/BluetoothGatt(18345): registerApp() - UUID=d2a73192-1369-4fda-ad16-f8a0b2535a85
D/BluetoothGatt(18345): onClientRegistered() - status=0 clientIf=6
D/BluetoothAdapter(18345): getBleEnabledArray(): ON
D/BluetoothGatt(18345): onClientConnectionState() - status=0 clientIf=6 device=XX:XX:XX:XX:08:92
D/BluetoothGatt(18345): discoverServices() - device: 7C:DF:A1:DB:08:92
D/BluetoothGatt(18345): onConnectionUpdated() - Device=XX:XX:XX:XX:08:92 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(18345): onSearchComplete() = Device=XX:XX:XX:XX:08:92 Status=0
D/BluetoothGatt(18345): configureMTU() - device: 7C:DF:A1:DB:08:92 mtu: 517
D/BluetoothGatt(18345): onConfigureMTU() - Device=XX:XX:XX:XX:08:92 mtu=23 status=0
D/BluetoothGatt(18345): setCharacteristicNotification() - uuid: 00002ade-0000-1000-8000-00805f9b34fb enable: true
D/BleMeshManager(18345): Data received -> mtu: 20, pdu: [65, 1, 0, -120, -30, -73, 36, 123, -105, -71, 90, 0, 0, 0, 0, 43, 69, 23, 29, -67]
D/BluetoothGatt(18345): setCharacteristicNotification() - uuid: 00002adc-0000-1000-8000-00805f9b34fb enable: true
D/BleMeshManager(18345): Data received -> mtu: 20, pdu: [-63, 110, 69, 122]
D/MeshManagerApi(18345): Received mesh beacon: SecureNetworkBeacon { KeyRefreshActive: false, IV Index: IV Index: 0, IV Update Active: false, Authentication Value: 0x2B45171DBD6E457A}
D/MeshManagerApi(18345): Secure Network Beacon authenticated.
D/MeshManagerApi(18345): Last IV Index: 0
I/MeshManagerApi(18345): Setting IV Update Active to: false
D/BluetoothGatt(18345): onConnectionUpdated() - Device=XX:XX:XX:XX:08:92 interval=24 latency=0 timeout=500 status=0
```

Below is the implementation of the connection to the provisioned node after provisioning:

```kotlin
fun provision(address: String, callback: (provisionedNode: ProvisionedNode) -> Unit) {
    provisioningService.provisionCallback = { selectedPeripheral, provisionedMeshNodeNode, unicastAddress ->
        val pnScanner = ProvisionedDevicesScanner(activity, meshManagerApi, bleManager)
        pnScanner.provisionedMeshNode = provisionedMeshNodeNode
        pnScanner.connectionCallback = SuccessCallback {
            Log.d("QBleMesh", "Connection callback was called.")
            startConfigurationProcess(selectedPeripheral, unicastAddress, callback)
        }

        bleManager.disconnect().enqueue()

        handler.postDelayed({ pnScanner.open() }, 2000)
    }

    val unprovisionedScanResult = unprovisionedScanResults.firstOrNull { it.device.address == address }
    unprovisionedScanResult?.let { result ->
        bleManager.connect(address) {
            Log.d("QBleMesh", "Connection done was called.")
            provisioningService.startProvisioningProcess(result)
[... 56 lines cut ...]
            isStarted = true
            true
        } catch (e: IllegalArgumentException) {
            isStarted = false
            false
        }
    }

    private fun stopScan(): Boolean {
        if (!isStarted) return true

        return try {
            BluetoothLeScannerCompat.getScanner().stopScan(this)
            isStarted = false
            true
        } catch (error: Exception) {
            false
        }
    }
}
```

Could you please help me understand why the characteristics are being overwritten and how to fix this issue?

Thank you!

ahmadshoh-orc, 2024-07-04:
I found the error. After disconnecting, I needed to first clear all characteristics in `BleMeshManager` and then reconnect to it. After adding this code, everything now works fine.

```kotlin
fun clearCharacteristics() {
  // Clear all cached characteristics and reset GATT instance
  meshProvisioningDataInCharacteristic = null
  meshProvisioningDataOutCharacteristic = null
  meshProxyDataInCharacteristic = null
  meshProxyDataOutCharacteristic = null
  gatt = null
}
```

closed 2024-07-05 by ahmadshoh-orc