nordicsemi/Android-nRF-Mesh-Library #586: Capacitor plugin supporting nRF-Mesh-Library

lebrislo, 2024-07-11:
Hi, 

Your nRF-Mesh-Library for Android and iOS are a great relieves when developing mobile application that communicate with Bluetooth Mesh devices. But in my case, I want to create an hybrid application using Ionic/Capacitor. Since I'm not using native solution, I have to implement a Capacitor plugin to interact with the nRF-Mesh-Library.

I'm wondering if the nRF team or anyone else plans to create the Capacitor plugin I need ? 

Or maybe it already exists. If you have any information about it keep me informed.

Thank you for your assistance !

roshanrajaratnam, 2024-07-13:
Hi Unfortunately we don't plan on creating such a plugin. However there are a couple of flutter wrappers developed by some of our contributors that you might be able to look up.

closed 2024-07-14 by roshanrajaratnam

lebrislo, 2024-11-19:
Hi, Ionic / Capacitor developers, I'm cooking the plugin right here : https://github.com/lebrislo/nrf-bluetooth-mesh-plugin