nordicsemi/Android-nRF-Mesh-Library #593: FOREIGN KEY constraint failed on loading of default mesh network

mpiffari, 2024-10-25:
---
In my KMM app, which is consuming the Android-nRF-Mesh-Library, after reception of MeshManagerCallabacks
- onNetworkUpdated
- onNetworkLoaded 

NRF Mesh library is getting a fatal exception accessing the DB.
I'm getting this exception during the verify initial startup and intialization of mesh manager API.

---
**Platform details:**

implementation("androidx.sqlite:sqlite-bundled:2.5.0-alpha10")

---
**Logs / Screenshots**
Exception stack trace:
`2024-10-25 18:17:01.980 22885-22942 AndroidRuntime app E  FATAL EXCEPTION: 
pool-3-thread-2
Process: app, PID: 22885
android.database.sqlite.SQLiteConstraintException: FOREIGN KEY constraint failed (code 787 SQLITE_CONSTRAINT_FOREIGNKEY)
at android.database.sqlite.SQLiteConnection.nativeExecuteForLastInsertedRowId(Native Method)
at android.database.sqlite.SQLiteConnection.executeForLastInsertedRowId(SQLiteConnection.java:961)
at android.database.sqlite.SQLiteSession.executeForLastInsertedRowId(SQLiteSession.java:790)
at android.database.sqlite.SQLiteStatement.executeInsert(SQLiteStatement.java:89)
at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeInsert(FrameworkSQLiteStatement.android.kt:38)
at androidx.room.EntityInsertionAdapter.insertAndReturnId(EntityInsertionAdapter.android.kt:100)
at no.nordicsemi.android.mesh.data.ApplicationKeyDao_Impl.insert(ApplicationKeyDao_Impl.java:112)
at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$insert$11(MeshNetworkDb.java:248)
at no.nordicsemi.android.mesh.MeshNetworkDb$$ExternalSyntheticLambda28.run(Unknown Source:4)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
at java.lang.Thread.run(Thread.java:1012)`

philips77, 2024-10-28:
Hi, let's wait for @roshanrajaratnam to come back.

Btw, you may check the database using Android Studio. Have a look at: https://developer.android.com/studio/inspect/database

iDistrax, 2025-06-26:
@philips77 Awesome thank you. Looking forward to the Kotlin release.

philips77, 2025-08-08:
Please forgive us delays caused by the vacation period.
We are focusing on the new library, but will try to keep the old one usable as well and fix bugs if necessary.
Regarding this issue, we need to wait until the main dev is back, sorry for the inconvienience.

roshanrajaratnam, 2026-08-27:
@mpiffari I would recommend migrating to the kotlin library because we focus more on it right now.  If you have a particular CDB JSON could you share it with me so I can try importing to see how this could be reproduced?