nordicsemi/Android-nRF-Mesh-Library #597: Stucking sometimes on Provision and sometimes on Getting Composition Data

HackHobby-Lab, 2024-11-11:
---

**Faileing to get composition Data or the provisioning**
I'm using nRF Mesh for quite long time now with my Mesh. But for past few months I couldn't work on it. Now after this time I got back to MESH, I tried to provison my nodes. But either sometimes I get stuck on Provision. Like it gets the APP keys and assign unicast address but on the Provision button does nothing. Or if I get past this problems after tries, I get stuck on Gettinh Composition data. It does not get the composition data.

**Expected behavior**
I should Search for Nodes, Identify them, Provision them, app should assign the app key and unicast address, and then The composition data should be taken.

**Platform details:**
 - Device: Redmi 9T,  Iphone(Tried on Iphone too)

**Logs / Screenshots**
[image]
[image]

roshanrajaratnam, 2024-11-13:
Thanks for reporting this issue, which version of the app are you using?

HackHobby-Lab, 2024-11-14:
> Thanks for reporting this issue, which version of the app are you using?

Hi Roshan!
I'm using App Version 3.3.4 for this case.

HackHobby-Lab, 2024-11-20:
> Thanks for reporting this issue, which version of the app are you using?

Any update?/

roshanrajaratnam, 2025-02-20:
Hi, I have not been able to reproduce this issue on our end. Are you still facing this issue?

roshanrajaratnam, 2026-08-27:
Closing due to inactivity

closed 2026-08-27 by roshanrajaratnam