nordicsemi/Android-nRF-Mesh-Library #599: Heatbeat publication message error

lebrislo, 2024-11-19:
Hi @philips77  and @roshanrajaratnam,

I have been using your Mesh library for building a Capacitor plugin that can control Mesh devices : https://github.com/lebrislo/nrf-bluetooth-mesh-plugin. For the app I'm developing, I need to know if my devices are online or not. I decided to use Heartbeat messages to complete this feature.

To implement this I'm sending ConfigHeartbeatPublicationSet to the provisionned node I need to know if it's online. The unicastAddress is the address of my node. The destinationAddress in my case, is my mobile phone **(the provisioner : 0x0001)**

```kotlin
fun sendConfigHeartbeatPublicationSet(
        unicastAddress: Int,
        destinationAddress: Int,
        count: Int,
        period: Int,
        ttl: Int,
        netKeyIndex: Int
    ): Boolean {
        if (!bleMeshManager.isConnected) {
            Log.e(tag, "Not connected to a mesh proxy")
            return false
        }

        val feature = Features()
        feature.proxy = Features.ENABLED
        feature.relay = Features.ENABLED

        meshManagerApi.createMeshPdu(
            unicastAddress,
            ConfigHeartbeatPublicationSet(
                destinationAddress,
                count.toByte(),
                period.toByte(),
                ttl,
                feature,
                netKeyIndex
            )
        )

        return true
    }
```

After sending this message I receive a ConfigHeartbeatPublicationStatus with statusCode = StatusCodeNames.SUCCESS(0x00). So far so good, but every time my node sends a Heartbeat message the Nrf-Mesh-Library is raising an error (See logcat)

```
2024-11-19 14:02:28.923  8231-8231  BleManager              no....mi.android.nrfmeshprovisioner  I  Notification received from 00002ade-0000-1000-8000-00805f9b34fb, value: (0x) 00-77-FC-1F-5A-7E-B9-7C-28-6B-AF-91-9C-F7-BC-39-D7-99-FD-9A-F0-BF
2024-11-19 14:02:28.924  8231-8231  MeshManagerApi          no....mi.android.nrfmeshprovisioner  V  Received network pdu: 0x0077FC1F5A7EB97C286BAF919CF7BC39D799FD9AF0BF
2024-11-19 14:02:28.931  8231-8231  BaseMeshMessageHandler  no....mi.android.nrfmeshprovisioner  V  Sequence number of received Network PDU: 29
2024-11-19 14:02:28.939  8231-8231  NetworkLayer            no....mi.android.nrfmeshprovisioner  V  TTL for received message: 4
2024-11-19 14:02:28.942  8231-8231  ValueChangedCallback    no....mi.android.nrfmeshprovisioner  E  Exception in Value callback
                                                                                                    java.lang.NullPointerException: Attempt to invoke virtual method 'no.nordicsemi.android.mesh.control.TransportControlMessage$TransportControlMessageState no.nordicsemi.android.mesh.control.TransportControlMessage.getState()' on a null object reference
                                                                                                    	at no.nordicsemi.android.mesh.transport.DefaultNoOperationMessageState.parseControlMessage(DefaultNoOperationMessageState.java:560)
                                                                                                    	at no.nordicsemi.android.mesh.transport.DefaultNoOperationMessageState.parseMeshPdu(DefaultNoOperationMessageState.java:78)
                                                                                                    	at no.nordicsemi.android.mesh.transport.BaseMeshMessageHandler.parseMeshPduNotifications(BaseMeshMessageHandler.java:156)
                                                                                                    	at no.nordicsemi.android.mesh.MeshMessageHandler.parseMeshPduNotifications(MeshMessageHandler.java:61)
                                                                                                    	at no.nordicsemi.android.mesh.MeshManagerApi.parseNotifications(MeshManagerApi.java:279)
                                                                                                    	at no.nordicsemi.android.mesh.MeshManagerApi.handleNotifications(MeshManagerApi.java:251)
                                                                                                    	at no.nordicsemi.android.nrfmesh.viewmodels.NrfMeshRepository.onDataReceived(NrfMeshRepository.java:496)
                                                                                                    	at no.nordicsemi.android.nrfmesh.ble.BleMeshManager$BleMeshGattCallbacks.lambda$initialize$0$no-nordicsemi-android-nrfmesh-ble-BleMeshManager$BleMeshGattCallbacks(BleMeshManager.java:121)
                                                                                                    	at no.nordicsemi.android.nrfmesh.ble.BleMeshManager$BleMeshGattCallbacks$$ExternalSyntheticLambda0.onDataReceived(Unknown Source:2)
                                                                                                    	at no.nordicsemi.android.ble.ValueChangedCallback.lambda$notifyValueChanged$0(ValueChangedCallback.java:184)
                                                                                                    	at no.nordicsemi.android.ble.ValueChangedCallback$$ExternalSyntheticLambda0.run(Unknown Source:6)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loop(Looper.java:246)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8653)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```

I have also tried to setup Heartbeat publication to my node by using your application based on nRF-Mesh-Library, but it raises the exact same error.

I don't if I can't set the provisioner (my application) as the destination of heartbeat message, but in the Bluetooth Mesh specification it's not contraindicated. I also don't know if I have to configure something prior to sending ConfigHeartbeatPublicationSet to my node.

Let me know if there's something I can do to fix my problem. Thanks !

Loris LE BRIS

roshanrajaratnam, 2024-11-20:
HI I did some debugging in to this and I noticed that we haven't added support for handling received heartbeat messages. Unfortunately I am a bit busy with some other mesh tasks. If you ever consider adding this feature we are open for pull requests :)

lebrislo, 2024-11-20:
Thank you for your answer,

Do you have any idea of the release date of the Kotlin version of the library ? Will it support receiving heartbeat message ?

As I hardly ever dug deep into Mesh transport layer, is the feature fast to implement or no ? Where in the library the process should be handled ?

roshanrajaratnam, 2024-11-21:
We don't have a date planned yet as it is still under development and I am still implementing some basic features in the reference app. Yes it will support handling heartbeat messages and we plan to add Server support i.e. allowing the phone to act as a node. I don't think it should take long to add that feature.

All you have to do is in the lower transport layer check the opcode and create a heartbeat message. You can use the `BlockAcknowledgementMessage` as a reference because a HeartbeatMessage is a ControlMessage that should be handled in the lower transport layer.

https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/37de84ddb24eeab044d041244b246ebf683cde22/mesh/src/main/java/no/nordicsemi/android/mesh/transport/LowerTransportLayer.java#L784-L795

lebrislo, 2024-11-29:
Hi, thank you for your explanations. Here's the PR for the implementation of receiving heartbeat messages #600 . This PR includes also support for Mesh model messages : GENERIC_POWER_LEVEL and LIGHT_CTL_TEMPERATURE_RANGE

closed 2025-01-17 by lebrislo